adafruit/Adafruit_Adalink #9: openocd issue with windows's path backslash.

hathach, 2015-08-27:
This line convert parameter to absolute path using native style https://github.com/adafruit/Adafruit_Adalink/blob/master/adalink/cores/nrf51822.py#L72 in case of windows it is ‘backward slash’. However, openocd.exe on windows still take the file parameter as it would on Linux, i.e ‘forward slash’.

DEBUG:adalink.programmers.stlink:OpenOCD response: Open On-Chip Debugger 0.9.0 (2015-05-19-12:09) Licensed under GNU GPL v2 For bug reports, read
        http://openocd.org/doc/doxygen/bugs.html
Info : auto-selecting first available session transport "hla_swd". To override use 'transport select <transport>'.
Info : The selected transport took over low-level target control. The results might differ compared to plain JTAG/SWD adapter speed: 1000 kHz Info : Unable to match requested speed 1000 kHz, using 950 kHz Info : Unable to match requested speed 1000 kHz, using 950 kHz Info : clock speed 950 kHz Info : STLINK v2 JTAG v23 API v2 SWIM v4 VID 0x0483 PID 0x3748 Info : using stlink api v2 Info : Target voltage: 3.205380 Info : nrf51.cpu: hardware has 4 breakpoints, 2 watchpoints target state: halted target halted due to debug-request, current mode: Thread
xPSR: 0xc1000000 pc: 0xfffffffe msp: 0xfffffffc
Error: couldn't open **C:UsershathachDropboxdafruiluetoothworkspaceAdafruit_nRF51822_FlasherAdafruit_BluefruitLE_Firmwaresoftdevices110_nrf51_8.0.0_soft
device.hex**

comment out the line  f = os.path.abspath(f), it will work (taking relative path). 

Maybe we should force forward slash always ? I am not sure how to do it properly (blindly replace ‘\’ with ‘/’ would cause issue with actual escape e.g space on linux).

tdicola, 2015-09-15:
Oh ack sorry missed this until now.  Ouch what a pain, it turns out OpenOCD's Tcl interpretor doesn't like backslashes and needs to have them escaped with a second backslash: http://sourceforge.net/p/openocd/mailman/message/28333473/  I just submitted a small fix to add this backslash escaping and gave it a quick test on windows.  It seems to load firmware fine and program it (using a nRF51822 as a test).  

Should just be able to do a git pull on the repo and then just for good measure run 'python setup.py develop' again (add sudo on Linux or OSX).  Let me know if you still see issues after this fix, thanks!

closed 2015-09-15 by tdicola

hathach, 2015-09-15:
It works perfectly, just tested :D, thanks

commit 9940f134 (2015-09-15): Fix #9 by escaping backslashes in OpenOCD path.