adafruit/Adafruit_Adalink #12: Command line argument for register reading

microbuilder, 2015-09-30:
It might be useful to have a command line argument that exposes the ability to read 8, 16 and 32 bit registers, so something like:

```
adalink stm32f2 -p jlink --readmem32 0x12345678
0xABCDF00D
```

tdicola, 2015-10-02:
Just committed a change to add these commands:
```
-r8 / --read-mem-8 ADDRESS
-r16 / --read-mem-16 ADDRESS
-r32 / --read-mem-32 ADDRESS
```
Each one takes an address which can be any 32 bit value, specified in decimal (123), hex (0x123), or octal (0123).  It will respond with the hex value, like '0xFF', printed to standard output.

Note that only one read mem command can be specified at a time--it's a little bit of an arbitrary limitation just to keep the meaning unambiguous (i.e. what order would results come back if multiple were specified, etc.).

closed 2015-10-02 by tdicola

microbuilder, 2015-10-02:
This is great, thanks for the addition.

```
$ adalink stm32f2 -p jlink -r32 0
0x200018F8
```

commit cb402b89 (2015-10-02): Resolve #12 and #13 by adding ATSAM21D18 core and --read-mem-8, --read-mem-16, and --read-mem-32 commands.